kwindrem/GuiMods #291: External transfer switch operation issue.

SaVok0887, 2024-12-28:
HI.
Im running gui modes on Cerbo which is operating 3.52.
I tried to use External transfer switch addon. digital input is setted to DI1 on Cerbo and is connected to auxiliary contact of grid contractor. input is changing status of External transfer switch status ( on grid/on generator.) but it is not influencing of AC1 input source. It stays all the time on Grid.
what can I check to find a bug and make this function work.
BR,
Sava
[image]
[image]
[image]
[image]

kwindrem, 2024-12-28:
You might check the log:

tail -40 /data/log/ExternalTransferSwitch/current | tai64nlocal

This should provide you some information about what that service is finding and doing.

SaVok0887, 2024-12-28:
Hi @kwindrem.
I checked log this is out come.
looks like exttransferswitch  part is not installed.
can it br an issue. as previously I was using external transfer switch function as additional script from setuphelper.
now when I try too install it it says that I cannot due to 3.52 cerbo os is in use.
how this function should be used now with a help of guimods only?
Thanks in advance.

This is log I get:
tail -40 /data/log/ExtTransferSwitch/current | tai64nloca                                                                                        l
2024-12-27 21:38:05.885562500 INFO:root:>>>>>>>>>>>>>>>> ExtTransferSwitch start                                                                                        ing (no version installed) <<<<<<<<<<<<<<<<
2024-12-27 21:38:05.948860500 INFO:root:Setting /Settings/TransferSwitch/GridCur                                                                                        rentLimit does not exist yet or must be adjusted
2024-12-27 21:38:06.077970500 INFO:root:Setting /Settings/TransferSwitch/Generat                                                                                        orCurrentLimit does not exist yet or must be adjusted
2024-12-27 21:38:06.210889500 INFO:root:Setting /Settings/TransferSwitch/GridTyp                                                                                        e does not exist yet or must be adjusted
2024-12-27 21:38:07.457571500 INFO:root:Setting /Settings/TransferSwitch/StopWhe                                                                                        nAcAvailable does not exist yet or must be adjusted
2024-12-27 21:38:07.651036500 INFO:root:Setting /Settings/TransferSwitch/StopWhe                                                                                        nAcAvailableFp does not exist yet or must be adjusted
2024-12-27 21:38:08.043632500 INFO:root:Setting /Settings/TransferSwitch/Transfe                                                                                        rSwitchOnAc2 does not exist yet or must be adjusted
root@MiWiFi-R4A-srv:~# /data/log/ExternalTransferSwitch/current | tai64nlocal
-sh: /data/log/ExternalTransferSwitch/current: No such file or directory
root@MiWiFi-R4A-srv:~# -sh: /data/log/ExternalTransferSwitch/current: No such file or directory
2024-12-27 21:38:07.457571500 INFO:root:Setting /Settings/TransferSwitch/StopWhe                                                                                        nAcAvailable does not exist yet or must be adjusted
2024-12-27 21:38:07.651036500 INFO:root:Setting /Settings/TransferSwitch/StopWhe                                                                                        nAcAvailableFp does not exist yet or must be adjusted
2024-12-27 21:38:08.043632500 INFO:root:Setting /Settings/TransferSwitch/Transfe                                                                                        rSwitchOnAc2 does not exist yet or must be adjusted

kwindrem, 2024-12-28:
The ExtTransferSwitch package has been obsoleted by GuiMods since v10.0. The functionality was moved into GuiMods so you no longer need ExtTransferSwtch. If you are somehow running both, this could be the cause of your issues. You would need to manually uninstall ExtTransferSwitch then uninstall and reinstall GuiMods. 

Unfortunately, the current version of SetupHelper is not compatible with the last version of ExtTransferSwitch (which is very old).

The easiest way  purge ExtTransferSwicth is to:
1)  remove ExtTransferSwitch it from the active packages list of PackagManager 
2) remove the package directory: rm -rf /data/ExtTransferSwitch
3) reload the firmware. To reload the same firmware version, switch the stored backup firmware then do an online update to the current version. If you are not on the current version, update via USB stick from the appropriate .swu file at:

https://updates.victronenergy.com/feeds/venus/

If these errors persist, get  back to me.

SaVok0887, 2024-12-28:
this is log form new setup so there was no old version of external transfer switch. I just installed setup helper and guimods.
but I will give a try to follow your instructions.
Thanks for support.

SaVok0887, 2024-12-28:
so. I just did reboot after I reinstall guimods and now it is working.
thanks

kwindrem, 2024-12-28:
Great. Glad it was simple.

closed 2024-12-28 by kwindrem