kwindrem/GuiMods #292: RpiTemperature

gainestr, 2025-01-21:
RpiTemperature package does not work with latest firmware version.  I realize this is not necessarily a GuiMods issue.

kwindrem, 2025-01-21:
I am not the author of RpiTemperature. You will need to contact the author for support. I searched GitHub and found this:

https://github.com/TimD1981/RpiTemperature

dsrawding, 2025-01-21:
You need to go into the setup file and change line 14 to:
source "/data/SetupHelper/HelperResources/CommonResources"

The path changed and that's why it's not working.

kwindrem, 2025-01-21:
What is in the setup file for RpiTemperature is correct on the main branch:

#### following lines incorporate SetupHelper utilities into this script
# Refer to the SetupHelper ReadMe file for details.
    
source "/data/SetupHelper/HelperResources/IncludeHelpers"

#### end of lines to include SetupHelper

The current version is v1.2

However, the setup script that's included in the latest tag is NOT CORRECT.

dsrawding, 2025-01-21:
Yes, I meant the setup script.  Thanks for catching that!

kwindrem, 2025-01-21:
My point is that you should NOT source CommonResources. Source IncludeHelpers instead or you won't get all the utilities things SetupHelper provides to the setup script.

Please move additional discussions to the RpiTemperature repo so the author has a heads up and can then work on a fix.

gainestr, 2025-01-21:
I appreciate the help.  My issue was that I was using "latest" branch instead of using "main" branch.  As soon as I changed the branch and re-downloaded and installed, everything worked fine.