kwindrem/GuiMods #293: GuiMods with Venus OS v3.60~20 /gui-v1 not loading anymore

kevbenjam, 2025-01-23:
When using GuiMods with Venus OS v3.60~20 /gui-v1 GuiMods isn't loading anymore. There seem to be some ["gui-v1 underwater improvements (built against a newer version of the graphics library)"](https://community.victronenergy.com/t/venus-os-v3-60-20-available-for-public-testing/13471).

mr-manuel, 2025-01-23:
@kwindrem you have to change `import QtQuick 1.1` to `import QtQuick 2`. This changed with GUIv1 `v6.6.0`.

kwindrem, 2025-01-24:
What firmware version switched to QtQuick 2. I need to put code in place to change the import line based on Venus OS version. Thanks.

mpvader, 2025-01-24:
Hi Kevin, v3.60~18

kwindrem, 2025-01-24:
Thanks. It'll take me some time to make this change. It will affect all my packages.

Looks like until then users will get a white screen !!!!!

mr-manuel, 2025-01-24:
Yes, but this time it shouldn't be such a big problem as last time, since the system is still reachable via GUIv2.

kevbenjam, 2025-01-24:
Thanks to y'all for figuring out the cause of the white screen so fast. For as long as the issue persists users are able to go back and stay at v3.60~17 or as said use the /gui-v2.

dsrawding, 2025-01-24:
I upgraded from v3.60~17 and went back and still get the white screen.  Even tried the stable release and then back with the same results.  Looks like it's gui-v2 for me for now.  Hopefully the effort to fix is not too bad.

kevbenjam, 2025-01-26:
@dsrawding and @kwindrem for me going back to v3.60~17 worked without any issues. I'm not having a white screen with that one and until GuiMods on v3.60~18 upwards is set up, it seems as if this is the best solution for now. The v3.60 beta versions seem to be on hold for now: https://community.victronenergy.com/t/venus-os-v3-54-1-available-for-public-testing/19958

kwindrem, 2025-01-26:
Yes, according to Victron the switch to the new QtQuick occurred with v3.60~18 so ~17 would be the old version and work with GuiMods.

I am in the process of making package installs more restrictive. In order for the package to install, the author must have tested it on a firmware version and added that version to a valid firmware version list. No longer will packages attempt to create file sets during the install process. This is an unfortunate change made necessary by the difficult recovery when the GUI presents a white screen making it impossible to access PackageManager. You will be able to bypass valid version checks but only from the command line.

kevbenjam, 2025-02-08:
@kwindrem and @mr-manuel is there anything that can be done to fix /gui-v1 when updating from v3.54 to v3.60~24 other than waiting for a new release of SetupHelper and / or GuiMods? Is there something I'm not applying, when updating or is it something that needs to be fixed from your side of things?

kwindrem, 2025-02-08:
The issue is changes in the stock code the prevents a simple update to GuiMods. I'm also not having a lot of bandwidth to devote to GuiMods and SetupHelper right now so making things work with the latest beta releases is going to take some time I'm afraid.

kevbenjam, 2025-02-08:
Thanks for your fast reply and ensuring that it's nothing I'm doing wrong when installing the newest betas.

kevbenjam, 2025-03-07:
@kwindrem Thanks for fixing the issue with the latest updates of [SetupHelper](https://github.com/kwindrem/SetupHelper) and [GuiMods](https://github.com/kwindrem/GuiMods). If someone is still experiencing a white screen with v3.60~25 you could also check if it is related to older versions of [dbus-serialbattery](https://github.com/mr-manuel/venus-os_dbus-serialbattery). I found out that this could also be the problem via the [Victron Community](https://community.victronenergy.com/t/v3-60-20-rpi-gui-v1-not-working/19856/3).

closed 2025-03-07 by kevbenjam