kwindrem/GuiMods #295: Problems with Venus OS v3.60~25

KAN-Milestone, 2025-02-23:
With the Venus OS v3.60~25 I can´t install GuiMods.
The error is "incomplete file set for v3.60~25.

/kim

kwindrem, 2025-02-23:
Yes, this is known. Recent changes in Venus OS are requiring significant changes to my packages and I currently have minimal time to spend on a rewrite. For now, if you want GuiMods, you will need to use an older beta if you have it or use the released version.

KAN-Milestone, 2025-02-24:
> Du får ikke ofte mails fra ***@***.******@***.***>. Få mere at vide om, hvorfor dette er vigtigt<https://aka.ms/LearnAboutSenderIdentification>

That´s OK.

Can you tell me what OS to use with what GuiMods version ?

Med venlig hilsen / Kind Regards

Kim Andresen
Senior Solution Engineer
Pre-Sales Nordics DK
Email: ***@***.******@***.***>
Mobile: +45 2552 9387

kwindrem, 2025-02-24:
GuiMods will work with v3.54 and v3.60~17 but not the more recent beta versions

kwindrem, 2025-02-25:
GuIMods v10.81 should now support v3.60~25

closed 2025-02-25 by kwindrem

KAN-Milestone, 2025-02-26:
Cool.

Med venlig hilsen / Kind Regards

Kim Andresen
Senior Solution Engineer
Pre-Sales Nordics DK
Email: ***@***.******@***.***>
Mobile: +45 2552 9387