kwindrem/GuiMods #297: Generator Manual Start - Run until all auto stop conditions are met - Stopped working

dinodeSimon, 2025-03-06:
At some point along the way this feature appears to have stopped working (at least for me). Previously it would run until the batteries were charged and then turn off the generator. The auto start / stop still works when the batteries hit a preset limit but the manual button to run until they are charged seems to have stopped working. Running the latest Cerbo firmware and Guimods.

When pressing manual start and selecting the all auto stop conditions mode it counts down to start the generator but then it never starts and just flicks from Manual Start - Active to Manual Start - Inactive straight away.

Is there any logging that could help troublshoot?

dinodeSimon, 2025-03-06:
Thanks. Firmware is 3.54 and GuiMods 10.81 (installing 10.82 now)

The battery soc condition hasn't been met when trying to start it. 

The condition works when it triggers based on the charge state hitting 30% automatically. 

Using the manual start for n minutes works fine as well. 

I'll check logs later today.

kwindrem, 2025-03-06:
I made some changes and it appears to work now. I have limited testing ability so give it a try and let me know.

dinodeSimon, 2025-03-06:
Tested 10.83 and it still didn't work. I'll SSH in later and look at logs. Appreciate your time

dinodeSimon, 2025-03-07:
`
/data/log/dbus-generator-starter# tail current
@4000000067ca2e152dc32bfc INFO:dbusmonitor:       com.victronenergy.system has device instance 0
@4000000067ca2e152f206a3c INFO:startstop:Generator0: Error getting Vebus service!
@4000000067ca2e170403272c INFO:startstop:Generator0: Error getting Vebus service!
@4000000067ca2e18052ad67c INFO:startstop:Generator0: Vebus service (com.victronenergy.vebus.ttyS4) found! Using it for generator start/stop
@4000000067ca2e182f88d0cc INFO:startstop:Generator0: Success getting (stoponac1) value, resuming evaluation
@4000000067ca2e182fa0f094 INFO:startstop:Generator0: Success getting (soc) value, resuming evaluation
@4000000067ca2e182fb98d5c INFO:startstop:Generator0: Success getting (acload) value, resuming evaluation
@4000000067ca2e182fd55a8c INFO:startstop:Generator0: Success getting (batteryvoltage) value, resuming evaluation
@4000000067ca2e183048b9b4 INFO:startstop:Generator0: Success getting (inverterhightemp) value, resuming evaluation
@4000000067ca2e18305c7864 INFO:startstop:Generator0: Success getting (inverteroverload) value, resuming evaluation`

Doesn't look like there is anything useful in there.....

kwindrem, 2025-03-07:
I did recreate the issue with battery voltage which I could manipulate to create a unsatisfied stop value and the change I put in as part of v10.83 did fix it. (I couldn't use SOC because my battery is fully charged and couldn't set the stop value greater than 100%,but battery voltage and SOC use the same logic: timed conditions, stop higher than start.)

I had a situation earlier today where I think one of my patched files didn't update on in install. I did an uninstall and reinstall and got the changes. Not saying that will fix your issue but it's a simple test.

I don't log the special manual start/auto stop condition so I didn't expect anything useful in the logs. I can add more log messages if we can't find the problem another way.

Do you know what firmware version this last worked on?

dinodeSimon, 2025-03-07:
Thanks Kevin. I just did an uninstall and re-install and it still has the same behaviour. 

I'm not sure how many versions ago it had stopped working but I hadn't updated for quite a while. 

Maybe it's time to change to the new UI and just do it with a timer. Or would this work with the new UI?

kwindrem, 2025-03-07:
None of the GuiMods user interface enhancements will be present in the New UI so you'd loose the manual start-auto stop functionality. Things that don't rely on the user interface like the external transfer switch would still work however. You just need to switch to the Classic UI for configuration then switch back to the New UI.

I'll do some more digging.

kwindrem, 2025-03-07:
I have not yet found any bugs in the code that would prevent manual start-auto stop from working. Please confirm that you have Auto Start enabled when you attempt to do this. If Auto Start is not enabled, the stop conditions will not be evaluated and the generator will never start.

kwindrem, 2025-03-08:
I found a bug in SetupHelper that was preventing proper updates to some files including the generator control. So none of my generator mods were actually being added to the system! I think this may explain why your system wasn't acting on the manual start / auto stop command. I didn't see this before because I was testing on v3.60~25 and SetupHelper was updating this version properly.

You should update SetupHelper to v8.29 then install GuiMods v10.84 which also adds additional logging of the manual start/auto stop function then retest.

dinodeSimon, 2025-03-08:
Cheers. Thats got it running again. Appreciate the assistance!

closed 2025-03-09 by kwindrem