kwindrem/GuiMods #298: Relay display name on the Gui-V2

Martial210, 2025-03-06:
Hello

The GuiMods on the Gui V1 works very well.
My question is about the GuiMods on the Gui-V2, renaming relays does not work.

I'm currently running version:
- 3.54 of Venus os
- 8.28 of StupHelper
- 10.81 of GuiMods

Is it possible to get some help ?
I can provide any additional details if needed
Thank you

Gui-V1
[image]

Gui-V2
[image]

kwindrem, 2025-03-06:
Sorry, GuiMods is not now and never will work with gui-v1 (aka New UI).

I will not be making any user interface changes to the New UI.

closed 2025-03-06 by kwindrem

drtinaz, 2025-03-06:
> Sorry, GuiMods is not now and never will work with gui-v1 (aka New UI).
> 
> I will not be making any user interface changes to the New UI.

I think you meant to say is not now and never will work with "gui-v2"