kwindrem/GuiMods #299: Additional temperatures not breaking out to second row.

morphias2004, 2025-03-31:
I have too many temperature sensors for one row on the main page. The README states that if there are too many, it should break out to a second ro.

Am I misinterpreting the README or is there setting I have missed?

[image]

I can manually slide the row to see the other temps.

Even keeping it as a single row that slides back and forth would be good.

kwindrem, 2025-03-31:
I can see how the ReadMe is confusing. What happens is that temps by them selves are displayed as a two-line high box. The same is true for tanks. If there are both temps and tanks, each are drawn as a single-line high box.

In all cases, if there are more than 5 temps or tanks, that row scrolls to show the additional ones. This is what you are seeing and explaining.

morphias2004, 2025-03-31:
Cheers for clarifying.

Is there a way to animate the temps to ping-pong across at bottom?

kwindrem, 2025-03-31:
no, not really

morphias2004, 2025-03-31:
No worries. Thanks for the prompt responses.