kwindrem/GuiMods #300: Alternator tile not displaying in GuiMods flow screen

davinsoftware, 2025-03-31:
GuiMods isn't displaying an alternator tile even though my alternator is active, is connected to the CerboGX, and the Cerbo has the alternator info available under the device list. 

Can anyone help figure out how to get the alternator tile to show up? It previously did show, last year, before I updated the Cerbo + GuiMods software. 

I have GuiMods "Flow overview" set to "GuiMods simple", but have also tried "GuiMods DC Coupled" and neither showed an alternator tile. I've tried enabling and disabling "Replace AC if inactive" but that makes no difference. Is there a setting for displaying or not displaying the alternator tile?  My Multiplus is set to "off" when motoring, so GuiMods is showing no tiles in the upper left or upper right. 

I'm on GuiMods 10.78 (latest is 10.85, but I see no changes in changelog related to alternator tile) and Victron 3.52.

kwindrem, 2025-03-31:
The simple overview should replace the AC in tile with Alternator if the former is not present and the latter is and you have Replace AC if inactive is enabled.

The DC Coupled overview should display an alternator tile if it is present.

Both rely on the system's alternator power to be valid so check to see if you have a valid power number for the alternator in the device list. If you have command line access, look to see if bus-spy reports an alternator in it's parent list and if there is an valid number for com.victronenergy.system /Dc/Alternator/Power

I created a dummy alternator service and the two flow pages displayed an alternator tile as described above.

The alternate device needs a dbus parameter of /Dc/0/Power.

I do not know of any changes that have been made that would prevent this from working.

kwindrem, 2025-04-02:
If there is no power number displayed in the device's menus, then it is pretty certain that there is no power number reported by the device. That's not something I have any control over.

davinsoftware, 2025-04-02:
Well, there has never been a power number in the device menu but I figured that just means Victron doesn't display it, not that it's absent. 

However I have some older screenshots from Sept 2023 showing GuiMods displaying our alternator regulator with no power number present. I feel like we had some past discussion about this - maybe it was here: https://github.com/kwindrem/GuiMods/issues/90#issuecomment-1730056003
In that thread you can see a screenshot with GuiMods showing the Wakespeed lacking a power number but summarizing Total Power as 0.0 W. (Note our alternator was in float then, so that may be why power wasn't reported or maybe Wakespeed just doesn't send it ever). 

The voltage, field and temperature of the regulator are still useful even if it's not outputting power. Ideally, any time the device is in "connected = yes" state the tile should be shown.

kwindrem, 2025-04-02:
I have found that /Connected is not a good indicator of whether the device is valid. It should, but have never seen it set to false!

I've checked my code and the Alternator details page uses voltage and current to calculate power if the power parameter is not valid. In your screen shots above of the device menu (from device list), current (or power) should be displayed on the same line as the voltage, so apparently Wakespeed is not providing either current or power. So I have nothing upon which to calculate power so the Alternator tile would always indicate 0 watts (or be blank) even if I showed it. 

Going back in some ole emails I found a mention indicating Wakespeed may not report current unless current is above 5 amps! This could be what we are seeing which is really too bad as 5 amps @12 volts is 60 wats which could be significant in an idle Victron system. This would need to be a change to Wakespeed code.

Historically the power value has been created by all devices in their "driver" code but I do see a few cases where power is calculated from voltage and current. The Alternator detail page already does this. I can do the same in the Alternator tile on the flow pages and also in systemcalc so the summed alternate power value will become valid any time there is a generator in the system. However this will not result in proper power values if Wakespeed does not report at least power or current and voltage.

davinsoftware, 2025-04-03:
Thanks. I don't think a 5-amp threshold is the issue in this case. (My Wakespeed was in bulk and outputting 60-70 amps). 

- I found an email I sent to Wakespeed in 2023. Apparently Wakespeed doesn't report current or power unless you have a dedicated shunt to measure alternator output (in addition to the default shunt which measures current going into the batteries). It's somewhat silly because current into the batteries will approximately equal alternator output in a single-charge system. But it is true that technically they don't know the alternator's output power. Although somehow they display alternator output power numbers in their own config/debug app (Android/iOS), which I suspect they approximate by multiplying field percentage with the rated alternator current (user inputted value).

- Regardless, previously GuiMods displayed our alternator tile even though it had 0 power / current reported. I don't know what changed to remove this capability. Perhaps it was a Victron core change. It would still be nice to have back because it gives us quicker access to useful data like the alternator field % and temperature. 

I can check if the alternator tile shows up in Victron stock UI (the new one). I will test more tomorrow or the next day when we have the engine on.

kwindrem, 2025-04-04:
Battery current and alternator current will NOT be the same unless there are no loads or other charging sources in the system which is unlikely. Battery current will be the difference between the sum of all charging sources (alternator, PV charger, other DC-DC chargers, inverter in charge mode, ...) minus the sum of all loads (DC loads, inverter while inverting, ...)

I do not know what has changed. Could be a change in Victron's driver or in Wakespeed code to remove the estimate it provides via its own UI.

Looks like Wakespeed sells a $70 shunt. Odd that it's an add-on to a $700 product !!!!! I would recommend installing the Wakespeed shunt so you have an indication of what's being delivered to the system.

I can make changes to systemcalc to create a system-wide alternator power if any alternator exists even if its power is not defined. I hesitate to do so as I may get complaints about a 0 power value when the alternator is obviously running.

It would be interesting to see what the New UI (aka gui-v2) does.

davinsoftware, 2025-04-05:
Yes, I understand battery current and alternator current aren't the same, I just wish the system could calculate alternator current for us since all the inputs are known to the CerboGX. (battery current - solar current + DC load). I know that's not the case in all systems though and out of scope for this bug. 

I've been in contact with Wakespeed, and they confirmed they don't provide current/power to the Cerbo unless you switch your battery shunt to be an alternator shunt. They only support one shunt. But if we moved it to measure alternator output, we would lose the key feature of the Wakespeed which is the ability to monitor tail current charge to the battery bank, allowing charge cycle control based on current (not just voltage), which is important for our LiFePo4 batteries. 

I tested Victron new UI and it does show the alternator tile for us.  With "-- W" but provides the benefit of knowing the alternator is active / working plus quicker access to its detail screen (voltage, field %, temperature).

I don't know how we lost that in GuiMods but we definitely had it shown in a similar way ("0 W" or "  W") before.   

[image]

[image]

closed 2025-04-06 by kwindrem

davinsoftware, 2025-04-07:
Thank you!!  I will update to 10.89 and check it out in a couple days.