kwindrem/GuiMods #302: White page on dashboard between relay and tank overview

felix1024, 2025-04-03:
Hi Kevin
Following your advice in Victron forum here again:

had to replace my Rpi3 on my RV. Upgrading from 3.32 to 3.54. I have local DSI touchscreen and use tank 140 as tank sensor.
I have GUImods installed, using several modules like RpiDisplaySetup, and others. RpiTemperature shows data of 5 Bluetooth coupled Ruuvi devices.

How can I get rid of of that white screen? I do not have a generator, but DC system with Multiplus, 2 Solar MPPT, BMV-712 Batterymonitor and a Orion XS booster.

Here output of
$ tail -50 /data/log/gui/current | tai64nlocal
2025-04-03 20:34:43.702788500 updateRelays
2025-04-03 20:35:32.260247500 ScreenSaver::timer 
2025-04-03 20:35:32.260253500 ScreenSaver::screenOff 
2025-04-03 20:35:55.330020500 VePlatformVenus::wakeupEvent 
2025-04-03 20:35:55.330175500 ScreenSaver::enable 
2025-04-03 20:35:55.330177500 ScreenSaver::screenOn 
2025-04-03 20:35:55.331158500 VeQuickView::mousePressEvent: wakeupEvent 

may need to increase logging level somehow?

More input:
I have a second Rpi3 on the workbench for testing purposes. it has the same configuration as the one in my RV. It was also running v3.32 perfectly. After updating to v3.54 the same additional white page in dashboard came in.

Here installed packages:

[image]

RpiShutdown is my own save shutdown/switch off script using a ONOFF SHIM hat.

BTW: Everything else in the RV works as expected so far. On the workbench no peripherals are connected!
If it would help digging in this problem could give you ssh access to the workbench rpi. (need to establish port forwarder on firewall)

Kindly
Felix

kwindrem, 2025-04-03:
Do you know which overview page is blank? I just fixed a problem with Tanks/Temps Dig Inputs overview that was blank for some OS versions. The relay page seems to appear properly. I've checked v3.54 and v3.60~53.

There is a dbus setting parameter /Settings/System/LogLevel  but don't know what affect that would have on GUI logging. Mine is set to 2. I will see all GUI messages including reports of problems with a specific page.

felix1024, 2025-04-03:
No the white page appears between the relay overview and the tank overview pages. Behavior is normal could wipe forward and backwards.

[image]

$ dbus -y com.victronenergy.settings /Settings/System/LogLevel GetValue
2
Changing number has no effect to log (tried 0,1,3,10)
See no log entry when changing page in dashboard

kwindrem, 2025-04-03:
All overview ages are loaded at boot and/or when switching between Menu and Pages so swiping to a new page probably won't show additional errors.

That screen shot is odd since the relay overview page is drawn to the left of the blank page yet the blank is labeled Relay overview.

With the blank page displayed, check the dbus Settings parameter  /Settings/Gui/DefaultOverview

felix1024, 2025-04-03:
Strange: on blank page:

# dbus -y com.victronenergy.settings  /Settings/Gui/DefaultOverview GetValue
'OverviewTanksTempsDigInputs'

The name on the white page depends from which page you swipe from tank -> tank from relay -> relay overview

kwindrem, 2025-04-03:
Yes, I just duplicated this. I was not seeing it because I was on firmware 3.60~53.

The blank page seems to be somewhat random.

felix1024, 2025-04-03:
OK when disabling "Show Tanks, Temps, Digital Input Overview" in GuiMods setting, the blank page disappears.
So the bug is probably in /opt/victronenergy/gui/qml/OverviewTanksTempsDigInputs.qml ?

I do not have any digital input activated

kwindrem, 2025-04-03:
I have a new version of GuiMods that should fix this. I found a problem in the generator overview page that seems to have fixed the problem. This latest problem was due to a typo in v10.87.

felix1024, 2025-04-04:
Good morning Kevin
Just installed Guimods 10.88 on Workbench Rpi3 and later on RV with great success!
Everything works perfect!
Thank you
Felix

closed 2025-04-04 by felix1024