kwindrem/GuiMods #303: IP43 Display as AC Charge in main screen

4ndiDev, 2025-04-11:
Is there a way that I havnt figured out yet to show shore charging from an IP43? I have no inverter on a sailing boat but would love to configure it that I have an AC in tile showing the loading. The IP43 also charges 3 batteries. One Service that is the main DC System with a Smart shant, the starter and an bow thruster / anchor battery. As the IP43 has the charge of them I would love to display it. Worst case, I could dig into code and see if I can figure out how to do it, but I'm totally new victrons code and I fear the learning curve would be quite steep. Can you point me in a direction?

When I bought the Cerbo, I wasn't aware it's neglecting the IP43 so much. I see that it has the data, and a nice overview screen suited for this setup would be lovely.

kwindrem, 2025-04-11:
I currently have code to replace the AC in tile with an alternator. There's an option in GuiMods under Display and Language that enables the replacement. Then if there is no AC input to the Multi/Quattro present and an alternator exists, that tile appears in the space provided.

I think setting the SmartShunt to DC Meter / Alternator would work. I could expand the logic to include AC Charger which is also a mode for the SmartShunt.

If you set your SmartShunt on the IP43 output to DC Meter / Alternator and it appears in the overview, then I'll look at adding AC Charger to the logic.

4ndiDev, 2025-04-11:
Ah. With the replace options the AC Charger showed up but kept being 0 even when there war charging. But that was on me. Works only for the first battery. My service was always the second. I switched them and now I see the shore charging! That’s very nice and exactly what I was hoping for. 

For the other thing, I might have been not clear enough in my description. My shunt is in battery surveillance mode and installed after the service battery. The IP43 charges three batteries and thus has the current voltage and I was only wondering if I can display those also on the overview screen. But I can now click on the AC charger and see the data in the details which is actually enough for now. 

Thanks a lot for the fast reply and keep up the good work. 
Andreas

kwindrem, 2025-04-11:
You want to keep a shunt for the battery in order to properly calculate state of charge. You'd loose that switching your only shunt to DC Meter mode.

I think what you have done is rename the shunt which then shows in tile for alternator 1 but only if there are additional alternators. AC Charger (or whatever name you've provided) does not show anywhere if there is only one alternator.

A second shunt is required since the IP43 has no VE.Direct or bluetooth connection to Cerbo.

4ndiDev, 2025-04-12:
[image]
[image]

The IP43 is connected to Crebo via VE.Direct cable. This is what it looks on display now. 
And in details I see all three batteries. Since I switched the main one to output 1 instead 2 it also displays the load from shore line when it's charging.

kwindrem, 2025-04-12:
Thanks. I stand corrected.

The DC Coupled flow overview has always shown the AC Charger tile. I am working on a change to the GuiMods Simple overview page to replace the AC Input tile with the AC Charger tile like I did with the Alternator tile. I'll let you know when I have something to test.

4ndiDev, 2025-04-12:
Thanks! Im open to test some stuff. I was looking a bit into the code to eventually change some general naming but I havn't grasp the general concepts yet. I assume you are not planing to support the new UI stuff as this looks more like a re-write?

kwindrem, 2025-04-12:
QML is pretty straightforward once you get the hang of it. I leaned by inspecting the existing code. Simple naming changes should be easy to identify and change.

At this point, I have no plans to make major changes to the New UI. Many of the changes I added with GuiMods are in the new UI any way. Currently, modifications must be made in two places: one for the local GUI and one for the remote ones and the remote one needs to be compiled into a single image. This makes multiple modifications very difficult: the changes would need to be made at the source level then compiled. The compilation must be done on a host computer or via GitHub's tools. I'm watching development to see what may be possible in the future.

kwindrem, 2025-04-12:
I've modified GuiMods Simple overview to show AC Charger tile in the AC In location if the AC input is not present. Alternator is shown in this location if AC In and AC Charger are not present.

closed 2025-04-12 by kwindrem

4ndiDev, 2025-04-13:
[image]

Looks good! Thanks.