kwindrem/GuiMods #304: White screen in Remote Console on GuiMods V10.91

FrannyannyB, 2025-04-13:
Hi there,

Just yesterday, my main display screen (GuiMods simple) went blank (white).  Is the V10.91 new?  I haven't changed anything else in my system, and it has been running and displaying just fine until yesterday.  I do have automatic updates turned on, and other Flow Overviews work.

How should I proceed?  From looking at other posts, this is usually an issue with the current version.  Should I wait for a new version?

Thanks,

Franny

kwindrem, 2025-04-13:
What firmware version are you running?

What platform? (Cerbo, Raspberry PI, etc)

kwindrem, 2025-04-13:
v10.91 changes GuiMods Simple to show an AC Charger if present and the AC in is not present.

GuiMods v10.87 fixed a bug which caused the Temps/Tanks/Dig In overview to be blank in some OS versions. So if you are running GuiMods version prior to that, I suspect that is the overview causing you problems. With this bug, the page title shown when cycling overviews is not correct which leads to confusion about which overview is actually the problem.

If you are not currently running v10.91, I would upgrade to it.

FrannyannyB, 2025-04-13:
Thank you for your response,
I am running V10.91. Is it very recent? I couldn't find a date as to when it was automatically installed.  The system was fine two days ago.  This just started yesterday. BTW, this GuiMods work you have done is amazing! Much better than even the most recent software from Victron.  I've come to rely on it.  Thank you for all your hard work on it!
If it isn't the version, do you have any suggestions as to what else I can try?
Thanks again.

FrannyannyB, 2025-04-13:
Sorry, Cerbo.

GuiMods: v10.91
SetupHelper v8.31
Cerbo: v3.30

What else can I get you?

Thanks!

kwindrem, 2025-04-14:
Thanks for the info. I found and fixed the problem. Only shows up in firmware prior to v3.40.

GuiMods v10.92 is available now with the fix.

closed 2025-04-14 by kwindrem

FrannyannyB, 2025-04-14:
You are amazing! Thanks so much!
We are on a boat, a sailing catamaran.  Heidi and Franny sailing.  I'm working through all the videos on the power system upgrade.  I'll let folks know about your GuiMods and how awesome it is!

Thanks, Franny