kwindrem/GuiMods #306: Relay Overview page  missing on Ver 3.60 release

RobDuthie, 2025-06-12:
Victron new release 3.60 has the relay overview page missing blank, every thing else is working OK.
What changes are required ?

RobDuthie, 2025-06-12:
Did the reinstall from the command line and from the touch screen, to see what happens, the relay page comes back OK, but if you do a reboot it is gone again Odd.?

kwindrem, 2025-06-12:
I am not seeing any issues with upgrade to v3.60 with GuiMods v10.94 and SetupHelper v8.33. Might be an issue coming from the previous released version. Qt Quick switched from 1.1 to 2 during the v3.60 beta.

I was coming from a recent v3.60 beta but not sure this makes a difference. I'll look into this further. Thanks for the report.

RobDuthie, 2025-06-12:
Yes i was using the older ver 3.50 with the RelayTile ver 1.1 that i was using with mods, so just by changing it to QtQuick 2 solved the problem. Thanks again for  advise.

kwindrem, 2025-06-12:
I am not able to reproduce this issue upgrading from v3.55 to v3.60.

Make sure you are running the latest SetupHelper. v8.28 adds a patch to change QtQuick imports from 1.1 to 2 so if you are running SetupHelper prior to v8.28 then that patch will not occur and many pages may not render.

If this does not make things install correctly, please provide more details about firmware versions before the upgrade. Logs might be helpful also.